thedalles77/USB_Laptop_Keyboard_Controller #2: Similar product?

0E800, 2019-03-16:
Was just wondering if I could purchase this as I have not the patience for soldering but would like to use a laptop keyboard as my main keyboard.

Ak5 Universal Laptop Keyboard Tester Testing Device Machine Tool USB Interface

https://www.ebay.com/p/Ak5-Universal-Laptop-Keyboard-Tester-Testing-Device-Machine-Tool-USB-Interface/5015695280?_trksid=p2047675.l2644

Any chance you might sell a few already assembled units?

thedalles77, 2019-03-20:
Sorry, you will have to do the assembly of the circuit board yourself. 
I don't know if the Ak5 keyboard tester will work for you but it's cheap enough that could give it a try.

0E800, 2019-03-20:
Thank you for your reply, I will give it a try.

0E800, 2019-03-21:
Upon doing more research, I found how the universal keyboard tester works. 
```
This device is easy to use.

1). Connect it to a computer by the USB cable.

2). Insert the keyboard cable into the corresponding connector.

3). Push the opening button, the device will sound 1 times, which means it starts working.

4). Press keys of the keyboard, if every keys sound, it means the keyboard is good, if not, the keyboard has problem. If the sound is long and can not stop, the keyboard might be bad or the cable is not installed correctly or firmly.
```

Do you think I could re-use the board for the connections and someone tie in your board so that it outputs through the USB and bypasses the speaker?

thedalles77, 2019-03-21:
Theoretically that could work but you would need to remove the controller
chip from their board so it does not get in the way of the Teensy when it
scans the keyboard. You would also need to jumper from their board to the
Teensy. It just seems like a whole lot of problems.

0E800, 2019-03-21:
Thank you for explaining that to me.
I need to suck it up and learn how to use a mutlimeter before I start frying more circuits.

Not sure if you are still pooling for group orders, if there is a small enough minimum I can afford.

Really cool of you to share this.

thedalles77, 2019-03-22:
Yes I'm still doing group orders. I can let you know in a few weeks when
they come in. A board is only $2.00 so it's cheap to give it a try. I'll
add your name to the list and when I have the boards, I'll get your address
from you.
Cheers

closed 2019-03-27 by thedalles77

thedalles77, 2019-04-02:
I just received 10 green circuit boards from JLCPCB in China. These boards
were only $7.24 for 10 boards (cheaper than the other company I've used). I
can mail you one board for $1.25 if you give me your U.S. mailing address.
After you receive the board in the mail, you can use this email address to
pay me via PayPal.
So far only one person has posted a picture of their finished keyboard
controller using the "I made it" link on my Instructable site (hint, hint).
Let me know your address.
Frank

0E800, 2019-04-09:
Apologies, I did not notice your update earlier.

In an effort to reduce the amount of spiderwebs and because I could not figure out how to send you a message in private on Github I am using 0bin.
https://0bin.net/paste/3DBkoakvMtN4yKcS#wqHnzhXEVOneOG6HNmDY6mz+QeYoSQa63lzWAQMjukx

May I purchase 2?  I usually break things at least once.
You may send any further correspondence such as paypal info to my email 0ld3nglish800@gmail.com.
Else, I will honor the email address on the postage.

Thank you sir. I have a variety of laptop keyboards and am looking forward to success.