thedalles77/USB_Laptop_Keyboard_Controller #3: T61/Teensy 3.2 Shopping List

riazarbi, 2019-03-25:
Hello,

Thank you for such an incredible resource, it's the first keyboard conversion project that actually looks possible for me (a NOOB) to take a crack at. 

The T61 Teensy LC `pdf` has a shopping list of components needed to replicate the project. The Teensy 3.2 version, however, does not. 

Would it be possible for you to add the components that would be needed to replicate the project? From reading the PDF I'm assuming you need - 

1. Teensy 3.2
2. RB-Spa-1385 teensy header kit
2. Lenovo_Scanner printed circuit board (get it printed yourself)
3. A T61 keyboard & trackpoint
4. WM6787CT-ND connector from digikey
5. 30 gauge wire
6. Soldering kit

Not sure if this is everything, though. 

FWIW I'll actually be using an x200 keyboard but the connector looks much the same.

WuerfelDev, 2019-03-26:
Good question (also noob here)
And please let us know how your process with the x200 is going

thedalles77, 2019-03-27:
Thanks for your interest in this project. Your parts list is only missing 10K leaded resistors for the trackpoint clock and data pull ups. I've updated the PDF files for both Lenovo T61 keyboard controllers so they give a complete parts list. I also added a picture to both PDF's that show how the T61 keyboard connector alignment tabs can be broken off due to using the WM6787CT-ND Digikey connector. The keyboard is still usable, it just takes more tries to get the contacts to align.

riazarbi, 2019-03-27:
Great, thanks @thedalles77 ! End goal is an x200 KVM crash cart with a lattepanda alpha instead of a Pi; will report back on the assumption that you'll find it interesting.

closed 2019-03-27 by riazarbi

thedalles77, 2019-03-27:
I definitely will find it interesting and others will too. I hope you post
a picture at the end of my KVM Instructable so we can all see it.