thedalles77/USB_Laptop_Keyboard_Controller #8: ThinkPad xx10, xx20, xx30 compatible?

exander77, 2021-04-07:
Hello, is there a version compatible with these keyboards?

thedalles77, 2021-05-25:
Sorry for the late response. 
I believe the xx10, xx20, and xx30 keyboards will need new connector boards. The same basic methods to decode the key matrix and send the USB HID codes will work the same as any other keyboard.

closed 2021-05-25 by thedalles77

diabl0w, 2024-11-20:
sorry to bump this, but I am currently trying to figure out if they xx20 uses the same connector as the t61. its driving me a bit nuts that i can't seem to find concrete info of this online. is there any chance you have found this out since the last post? I am trying to build a "lapdock" and this project seems perfect for me

diabl0w, 2024-11-20:
following up on my own comment... im 99% sure xx20 xx30 uses the "JAE CONN40A-1-U1GP" onboard connector. I don't have any pcb schematic drawing skills, not sure if anyone would be interested in taking on creating a pcb schematic for it, but it would be MUCH appreciated

diabl0w, 2024-11-20:
on the otherhand this: 
```
https://gitlab.com/XenGi/tpwkbd
```
 and this: 
```
https://img.alicdn.com/imgextra/i1/189306225/O1CN017V6P9H1vr6G0rtfLY_!!189306225.png 
```
seem to imply that the t61 connector would work for xx20/xx30 keyboards

thedalles77, 2024-11-20:
I'm a little confused by your xx20/xx30 nomenclature. If you're referring
to a T420 or T430 then yes these keyboards use the same connector as the
T61 and you can use the same circuit board that is described in my
instructable.

diabl0w, 2024-11-20:
> I'm a little confused by your xx20/xx30 nomenclature. If you're referring to a T420 or T430 then yes these keyboards use the same connector as the T61 and you can use the same circuit board that is described in my instructable.
> […](#)

I apologize, perhaps I'm confusing myself, however I was following the same nomenclature as the title of this issue. xx20 referring to x220/t420/etc and xx30 being x230/t430/etc. 

In my case, I have an x220 keyboard. Would you say I should be fine using the t61 board from this repo?

thedalles77, 2024-11-20:
Yes I be

thedalles77, 2024-11-20:
> ID: <thedalles77/USB_Laptop_Keyboard_Controller/issues/8/2489108054@

Yes I believe it will work

diabl0w, 2024-11-20:
> Yes I believe it will work

Awesome, thank you!