thedalles77/USB_Laptop_Keyboard_Controller #11: Elan touchpad

r3l4x-pt, 2022-01-13:
Hi. Im trying to use a Elan 33200v-3600 touchpad with a teensy 2 ++. Im using the Touchpad_3p2.ino example. 

The problem i have is the cursor will keep going until i tap the touchpad. 

For exampe i swipe left and until i tap the touchpad it wont stop going left. 

Also to make it "work" i modified the code in touchpad_init() function from 

if (tp_read() != 0xfa)  to  if (tp_read() != 0xfffffffa)

If i remove tp_read  it will also work but if i dont remove or add the extra f it will not work.

r3l4x-pt, 2022-01-15:
So i ve tried the Touchpad_Stream_mode.ino and that script works perfectly without any modification. I messed arround with both scripts and unless the touchpad is in stream mode the mouse will keep moving until i tap it.

thedalles77, 2022-01-27:
I had this same problem with an Elan touchpad as you may have read in the comment section of my Hackster.IO tutorial. Switching to stream mode fixed it for me too. It was someone else's touchpad so I don't have access to it anymore. 
If I ever get another Elan touchpad that has this problem when in "remote mode", I have an idea to try out. The idea is:
After reading the X and Y values, I will write zero's back to these registers in the touchpad. I don't know if this can be done but I'll give it a try.

r3l4x-pt, 2022-01-28:
for the moment its "fixed" with this code:

char oldmx;
  char oldmy; 

if (mx == oldmx) {
    mx = 0x00;
  }
    if (my == oldmy) {
    my = 0x00;
  }

// send the x and y data back via usb if either one is non-zero
    if ((mx != 0x00) || (my != 0x00)) {
      Mouse.move(mx,my);
    oldmx=mx;
    oldmy=my;
    }

i think the problem is with the overflow bits. on you code you have 0x80 and 0x40 and i get the some random number. i used Serial.println(mstat, HEX); to see what i was getting

thedalles77, 2022-01-28:
Thanks for your work-around code to "fix" the problem. Perhaps Elan has redefined bits D7 and D6 in the status byte and it's no longer for overflow. The code to look at the overflow bits in the status byte can be removed. The x and y values should not overflow as long as you are scanning the touchpad fast enough.  If I ever find another Elan touchpad to test, I'll try your code and look at the overflow bits. I'll also try writing zero's to the touchpad to clear the x and y registers after they've been read. Let me know if you find out anything else. 
Thanks again for keeping me updated.

closed 2022-02-11 by thedalles77