thedalles77/USB_Laptop_Keyboard_Controller #13: Where can I get a hold of your circuit board?

Bergum, 2022-02-15:
I think i have gone over most of what you have written, but i can't find either where I can buy a board, or find what I need to have one manufactured it...

I just have to say that I love the work you have done with the kwm.

thedalles77, 2022-02-19:
The unpopulated circuit board can be manufactured at any PCB fabrication company. I often use JLCPCB.com if I can wait a month for economy shipping from China. OSHPark.com in the U.S. is more expensive but only takes about 12 days to ship within the U.S. All any board shop needs is the circuit board file in a Zipped Gerber format which I have provided in the Keyboard_Scanner_Eagle_Layouts folder at my repo. There's lots of files but the one that will work with a Teensy LC on one side or a 3.2 on the other side is called "Keyboard_Scanner_LT2_2020-06-09.zip". Step 6 of my [Instructable](https://www.instructables.com/How-to-Make-a-USB-Laptop-Keyboard-Controller/) has more details but let me know if you still have questions.

Bergum, 2022-02-19:
Thank you. Was thinking on my project right now..

closed 2022-02-19 by Bergum